Duet3D/DuetWebControl #308: G-code input must not do greedy capture from history

tdcox, 2020-12-14:
Duet Web Control 3.2.0-beta4

Currently, when a user starts typing G-code into the input field on the dashboard page, an arbitrary match is made with a previous command string from the history. Pressing Enter executes the match from the history, NOT the command typed by the user. Since the match sometimes appears to be greedy, if the user types `G0 X0 Y0` and the history contains `G0 X0 Y0 Z0` then an additional Z move will be executed, crashing the tool into the work.

[image]

Similarly, if, with tool 0 loaded, the user types `T1` but a prior `T1 P0` is in the history, this will be executed preferentially, changing the state of the loaded tool silently, without actually replacing the tool. This can lead to a subsequent tool crash, or if the command `T-1` is issued, tool 0 will be smashed into tool 1.

To make things worse, the matching appears to be arbitrary, in that sometimes an item from the history is pre-selected and at other times it is not, even though matches exist. This leads to entirely unpredictable system behaviour.

To minimise the risk of damage or injury, the system should NEVER execute any action other than that explicitly commanded by the operator, so pressing Enter should always execute the command typed into the buffer, not the suggested match from the history. Selection of history items should require an explicit action to load the command into the buffer and another specific action to execute it.

tdcox, 2020-12-14:
Actually, this is worse than I thought. What is arbitrary is the presence or absence of the highlight indicating the match being made in the history. I just repeated typing `G0 X0 Y0`, got no indication of a match with anything in the history, but the system still executed `G0 X0 Y0 Z0` on Enter.

chrishamm, 2020-12-15:
I am sorry but I am having trouble reproducing this. When you use arrow keys to select an item, you can use Enter to send it. If there is an entry `G1 X100 F6000` in the suggestions, I can see it when I type `G1 X1` but when I press Enter, `G1 X1` is sent as expected. Or I can select it using Up/Down and then send that selected item with Enter. If I don't want to use a previously selected item, I can press Escape, the drop-down closes, and the selection is no longer active.

Please share more details like the browser you are using and detailed instructions how this problem can be reproduced.

tdcox, 2020-12-15:
Using Chrome 86.0.4240.183 from MacOS 10.15.7, the history search box drops down as soon as the text field gets focus and becomes active, sometimes with visual feedback highlights and sometimes silently. Text entry appears to go to the search widget in preference to the text box itself, so typing is actually providing search filter input and Enter selects from the search box.

zombiRon, 2020-12-17:
I too have experienced this, I too suspected a browser issue but have observed the same behaviour with Safari, Chrome, Firefox and Microsoft Edge, on Windows and macOS. To confirm the behaviour, the suggested history item overrides the typed command if it begins with the same characters. I have resorted to using telnet instead of the console within DWC.

chrishamm, 2020-12-17:
I will try to reproduce it again later today. In the meantime you can still
turn off auto complete to prevent the suggestions from popping up.

chrishamm, 2020-12-18:
This will be fixed in 3.2-RC1.

closed 2020-12-18 by chrishamm