Duet3D/DuetWebControl #310: Laser Mode displays too much

cdedwards, 2020-12-18:
Is there any chance that when in laser mode, some of the display's that would be relevant for a 3d printer but not a laser be removed. Examples would include the heaters, Tools, temp chart. Temp chart could be configurable since, I might actually hook a thermistor up to keep and eye on coolant temperature. Same with DWC config with Tool Temp's, Bed Temp's and chamber temps all being shown. With a laser I'm not concerned about these at all. We might want to display temp for cpu and drivers.

Sindarius, 2020-12-18:
There is an experimental version of the DWC that had a CNC targeted dashboard. You can check it out here.  https://github.com/Duet3D/DWC-CNC/releases

chrishamm, 2022-01-28:
Closing due to inactivity.

closed 2022-01-28 by chrishamm