Duet3D/DuetWebControl #322: Feature Request (Customize theme colors)

s7a73farm, 2021-01-15:
Would love to have the ability to pick theme colors like the gcode viewer plugin. But the theme color of the entire web gui.

chrishamm, 2021-01-19:
In any case I wouldn't want to make this a core feature; a third-party plugin would be much more appropriate. Check out https://vuetifyjs.com/en/features/theme/ if you want to see how the colors can be changed. I have no idea if they can be easily changed during runtime though.

joshenders, 2021-02-11:
@chrishamm, I have a use case which you may not have thought of: many users have multiple printers (printer farm) and want to easily differentiate between them (imagine the frustration of accidentally clicking stop on the wrong printer mid print!).

I was considering adding support under the appearance section in the settings.

I'll draft a PR if you're open to it?

chrishamm, 2021-02-11:
DWC 3.3 will get multi-machine support so I am not sure if it's the right time to write a PR for this. The backend has supported multiple machines since DWC2 but I haven't had time to add the necessary controls and to test everything yet. But now that users can write third-party plugins (which will be fully unlocked in v3.3) I think it's a good idea to do that next.

joshenders, 2021-02-11:
In the meantime, I've written a userscript for anyone immediately needing this functionality: https://github.com/joshenders/userscript-duetwebcontrol-colors

dans98, 2021-02-13:
I'd like to see this as well. 

I'd also really like to have the ability to adjust font size and spacing/padding in some way.