Duet3D/DuetWebControl #325: [Enhancement]

CabalSoul, 2021-01-19:
The temperature chart is visually appealing ,but with a higher temperature Hotend (550°C)  all  other sensors are scaled to 550 units. Most Sensors DHT, Air Temp, Bed ,  have their max value at 80 and it makes the chart unusable for everything except the high temp hotend. 

Being able to see the numerical values of the sensors in the  chart or nearby without needing to switch between Tools and Extra would improve IMO the usability even further.

wildblue, 2021-03-20:
With a chamber heater it makes the chart completely useless. Please fix this as soon as possible.

NikAWing, 2021-12-27:
This, and maybe this, too: Please add a feature that allows magnifying the temp chart, either temporarily increase the size in browser or open as new window.
it would make it much easier to see small temperature differences/fluctuations that sometimes are helpful to see