Duet3D/DuetWebControl #333: Feature Request: Implement "Slicer Override" feature into the filament config.g file

Nathan116, 2021-02-20:
When I first started using DWC a few months ago I was under the assumption that the filament config.g file was sort of like a "slicer setting override" file where you could modify things like temperature settings for you various different filaments. I have since found out that it is really only useful for settings like different temperature PID values and acceleration values; not the most useful feature.

I was wondering if there was a way to instead convert the config.g file, or maybe implement another file, that could be used to override slicer settings such as temperature and extrusion multiplier, and maybe even things like retraction and speed settings. This would allow the same Gcode file to be printed in different filament types and would allow you to have just 1 generic slicer profile that you would then just select the filament you wanted to print the sliced model in. This would greatly improve file management and would mean you wouldn't need to have several different gcode files for the same model, just different plastic types.

chrishamm, 2021-02-21:
You can achieve this behavior by putting [M703](https://duet3d.dozuki.com/Wiki/Gcode#Section_M703_Configure_filament) into your tpostX.g macro file so config.g file of your loaded filament runs whenever you change/activate the nozzle.

In the future please use the [forum](https://forum.duet3d.com) for support questions.

closed 2021-02-21 by chrishamm

Nathan116, 2021-02-21:
> In the future please use the [forum](https://forum.duet3d.com) for support questions

I apologise. I had asked a few people and they said that the feature didn't exist and that I should make it a feature request.