Duet3D/DuetWebControl #334: To send commands via the input field at the very top, the button has to be clicked 2 times.

well-engineered, 2021-02-21:
Hi!

To send commands via the input field at the very top, the button has to be clicked 2 times.

I noticed, that I have to click on "send" two times, before the command is actually sent to the printer.

If I click somewhere else to let the cursor in the inputfield disappear, and then click on the send-button, it works at the first click.

I think, it has something to do with the input-field being "active" or such.

chrishamm, 2021-02-21:
Please share some more details:

- What browser are you using?
- Do you have DWC 3.2.2 installed?

well-engineered, 2021-02-21:
Browser: Safari on MacOS Big Sur
DWC 3.2.2 installed but "Setting -> General" shows "Duet Web Control 3.1.1". 
I tried to install 3.2.2 one or two times again, but still 3.1.1 is showing. Maybe a bug, maybe I didn't upgrade it right?

well-engineered, 2021-02-21:
Here is a video, 2 times with clicks, the last time with "pressing Enter".

https://youtu.be/1EvXyKaXLQ0

chrishamm, 2021-02-21:
Thanks, I'll check if I can reproduce it. Try reloading the page via Cmd+Option+R and check if it shows v3.2.2 then, else try clearing your browser cache.

chrishamm, 2022-01-28:
This should be fixed already. If it isn't, please create a new issue on the forum.

closed 2022-01-28 by chrishamm