Duet3D/DuetWebControl #338: Prusaslicer direct upload with folders

nofuturekid, 2021-03-06:
There seems to be a problem with uploading from Prusaslicer when there are folder names the filename.

[image]

Duet (standalone)
- If the folder already exists everthing is ok
- If the folder is not existing then upload fails with: 
[image]

Duet (with SBC)
- Upload never fails. The file is not put into the correct folder but in the main folder a file with "Folder/Filename.g" is created.
- No matter "Folder" exististed prior uploading or not

chrishamm, 2021-03-06:
It isn't a bug in DWC so I will close this issue. That problem should be reported on the PrusaSlicer GitHub page instead.

closed 2021-03-06 by chrishamm

nofuturekid, 2021-03-06:
If DWC would behave the same while in standalone and while running on SBC I would agree.l, but it's not.