Duet3D/DuetWebControl #340: Feature Request: Open selected G-Code command reference

kauedg, 2021-03-10:
When editing a `.g` file the "GCode Reference" button is shown on the top bar. When clicked, it opens Duet's GCode reference and then the user searches for the desired command information. It's very helpful as it avoids keeping the documentation bookmarked. But I'd like to propose an extension of it's funcionality.

Would it be possible to make it open the reference documentation of a selected command in the editor?

Example:

I'm editing this file and need the reference documentation for the `M566` command. I select it and click the "G-code reference" button.

[image]

This action would take me to this URL:
>https://duet3d.dozuki.com/Wiki/Gcode#Section_M566_Set_allowable_instantaneous_speed_change

This would save the trouble of searching for the command reference manually.

wildblue, 2021-03-20:
I would Like to See this implemented, top.

T3P3, 2021-04-28:
We are reviewing options for the documentation gcode page overall so it would make sense to defer this feature until this is complete.