Duet3D/DuetWebControl #350: program help

cdedwards, 2021-03-30:
So I'm attempting to continue the DWC-CNC project and I've run into a snag. Yes I know this is not really the place to ask questions, but 
Anyways, I'm attempting to conditionally include the Filaments menu depending on what mode we are in. In Filaments.vue I've added the following

					Filaments: {
						icon: 'mdi-radiobox-marked',
						caption: 'menu.files.filaments',
						path: '/Files/Filaments',
						condition: () => (this.machineMode === MachineMode.fff)

This is the same code used in isFFF() or isFFForUnset() functions so we know it works. This code will always show the Filament's menu. Adding a ! in front of () will turn off the filaments menu. It does not do so when changing with M451/M452/M453 however. I seem to be missing some logic. Yes I tried using the isFFF() function, but that is a whole other ball of worms. basically I'm using the same method that is used in Display.vue with it being conditionally displayed in the left side bar.

BruceRoyce, 2021-03-30:
Without inspecting the complete code, it is difficult to diagnose, however, my guess is that your code is being run once. It reads the this.machineMode and sets the boolean result of your condition only once (that -let's say- is true when page first loads, so the menu shows). Whereas there must be trigger implemented to rerun the condition one M452 or M453 command are issued, so the code can pick up the change in the condition. You may write a listener to monitor the change in the MachineMode, or set a handle where M451/452/453 are issued. 
Just thinking aloud though :)

cdedwards, 2021-03-30:
I think you are correct now that someone has voiced it. I'm not sure how todo what's needed at the moment then. I don't see any listener setups with in the current code I'll have to look into that though.

BruceRoyce, 2021-03-30:
One slightly lazy solution can be passing the trigger to the user. By placing the option to change the machine mode on a different page, or on a pop up menu (like a "Machine Setting" page, etc.), or similar, that forces the user to click a button or any kind of a known interaction in order to confirm machine mode change, you can then use that click/trigger to redo the conditional test and subsequently reset the menu display.

cdedwards, 2021-03-30:
there are no buttons for this. typing M452 into the MDI will force a change from FFF mode to Laser. M453 will goto CNC mode and M451 will revert back to FFF

carloslema, 2021-04-07:
@cdedwards curious, are you building from DWC-CNC repo?

cdedwards, 2021-04-07:
yes I am looking at the DWC-CNC repo.

chrishamm, 2022-01-28:
This is actually a good idea, I'm going to add this to the next version. The right solution would've been

```
...
import store from '@/store'
import { MachineMode } from "@/store/machine/modelEnums";
...
		registerRoute(this, {
			Files: {
				Filaments: {
					icon: 'mdi-database',
					caption: 'menu.files.filaments',
					path: '/Files/Filaments',
					condition: () => store.state.machine.model.state.machineMode === MachineMode.fff
				}
			}
		});
```

closed 2022-02-10 by chrishamm