Duet3D/DuetWebControl #353: ATX button always OFF when connecting (even if the printer is ON)

Fulg, 2021-04-04:
Hello, it seems that the ATX power box default is always "off" instead of the current state. Easy repro:

- In the Dashboard, click on ATX ON.
- Refresh the web page (F5).
- Notice that ATX is now OFF.

My printer actually uses PSON and I have `M80` in my `config.g` so I was surprised to notice the button in the wrong state (the printer was powered so I knew DWC was lying!).

The buttons work just fine, it is just the initial display that is wrong, it does not reflect the current state (it is always assumed OFF when first connecting).

DWC 3.2.2, RRF 3.2.2.

dc42, 2021-04-07:
Which Duet are you using? There is a known issue with the reporting of PS_ON on the Duet 3 Mini with firmware earlier than 3.3beta3.

Fulg, 2021-04-07:
I indeed have a Duet3 Mini5+. I should have confirmed on Duet2 before opening this issue... :(

closed 2021-04-08 by dc42