Duet3D/DuetWebControl #358: M117 message not always displayed after being discarded

khenderick, 2021-04-16:
With e.g. `M117 "foo"` it is possible to show a message containing `foo` in DWC. This message is persistent, but can manually be discarded. 

If the message is discarded, and you execute the same command (`M117 "foo"`) again, the message is however not shown again. Only when displaying another message first (e.g. using the command `M117 "bar"`, showing `bar`), the message `foo` can be displayed again using `M117 "foo"`.

While this makes sense when displaying the message on an LCD screen (the message there is always persistent, no need to show it again), it can be counter-intuitive in DWC as a user can discard messages.

chrishamm, 2022-01-28:
I won't change the basic behaviour of this - if you send `M117 "foo"` twice and previously dismissed the old identical message, IMHO it should remain hidden. If you want to display regular notifications, use M118 instead of M117. To work-around this, you could remove the message (`M117 ""`), wait a moment (`G4 S1`), and display it again (`M117 "foo"`).

An alternative may be to make custom messages "uncloseable" using a setting in DWC. Would that work for you?

chrishamm, 2022-02-10:
Closing due to inactivity.

closed 2022-02-10 by chrishamm