Duet3D/DuetWebControl #361: Emergency stop button can get covered by too many notifications

mfs12, 2021-05-10:
When to many notifications show up the emergency stop button gets covered by them. 

As this button is important safety feature and should be always quickly accessible i would expect this doesn't happen.

Moving the button to another position could solve this issue easily.

BruceRoyce, 2021-05-10:
The fastest and easiest way to fix this (at least temporarily) is to increase the button z-index so it can stay on the top of everything else.

BruceRoyce, 2021-05-10:
I applied the same interim fix as I described above and pulled a request here:
[](https://github.com/Duet3D/DuetWebControl/pull/362)
Hopefully someone will pick it up :)

natthapolvanasrivilai, 2021-11-20:
I had a couple hard crashes due to the E-Stop button being blocked by M292 notification, or even worst in the PanelDue notification.

chrishamm, 2022-01-27:
This is going to be addressed in the next version.

chrishamm, 2022-02-10:
Now changed in 3.4-rc1.

closed 2022-02-10 by chrishamm