Duet3D/DuetWebControl #369: Intermittent tool info not being reported

WheelerCC, 2021-06-06:
**Issue:** When on the home page (dashboard), sometimes when the page loads the heater info just doesn't load and remains N/A. I've noticed this since upgrading web control from 3.2.2 to 3.3.0-rc3. 

[image]

**Steps to reproduce:** Chrome > navigate to 192.168.0.212 > look at "tools" section of the dashboard page.
**Steps to resolve:** Profusely refresh the page, eventually it goes back to normal behaviour.

Not sure the best way to provide some kinda debuggable info, but just let me know what's useful for you and I'll try to provide it.

```
M122
=== Diagnostics ===
RepRapFirmware for Duet 2 WiFi/Ethernet version 3.3RC3 (2021-05-26 13:47:14) running on Duet WiFi 1.02 or later
Board ID: 0JD0M-9P61A-J8PSJ-6J1FJ-3SD6M-98R6T
Used output buffers: 3 of 24 (22 max)
=== RTOS ===
Static ram: 23876
Dynamic ram: 76052 of which 256 recycled
Never used RAM 11376, free system stack 96 words
Tasks: NETWORK(ready,114.9%,221) HEAT(delaying,2.6%,330) Move(notifyWait,11.0%,307) MAIN(running,178.9%,441) IDLE(ready,2.9%,29), total 310.3%
Owned mutexes: WiFi(NETWORK)
=== Platform ===
Last reset 54:03:13 ago, cause: software
Last software reset at 2021-06-04 12:14, reason: User, GCodes spinning, available RAM 13728, slot 2
Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x0041f000 BFAR 0xe000ed38 SP 0x00000000 Task MAIN Freestk 0 n/a
Error status: 0x00
Aux0 errors 0,0,0
Step timer max interval 0
MCU temperature: min 29.4, current 36.7, max 37.8
Supply voltage: min 23.8, current 24.2, max 24.4, under voltage events: 0, over voltage events: 0, power good: yes
Heap OK, handles allocated/used 99/2, heap memory allocated/used/recyclable 2048/532/464, gc cycles 2
Driver 0: position 43400, standstill, SG min/max 0/1023
Driver 1: position -39400, standstill, SG min/max 0/1023
Driver 2: position 39306, standstill, SG min/max 0/1023
Driver 3: position 0, standstill, SG min/max 0/1023
Driver 4: position 0, standstill, SG min/max not available
Driver 5: position 0
Driver 6: position 0
Driver 7: position 0
Driver 8: position 0
Driver 9: position 0
Driver 10: position 0
Driver 11: position 0
Date/time: 2021-06-06 20:45:50
Cache data hit count 4294967295
Slowest loop: 139.24ms; fastest: 0.14ms
I2C nak errors 0, send timeouts 0, receive timeouts 0, finishTimeouts 0, resets 0
=== Storage ===
Free file entries: 9
SD card 0 detected, interface speed: 20.0MBytes/sec
SD card longest read time 4.0ms, write time 62.0ms, max retries 0
=== Move ===
DMs created 83, maxWait 139857197ms, bed compensation in use: mesh, comp offset 0.000
=== MainDDARing ===
Scheduled moves 24932, completed moves 24932, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
=== AuxDDARing ===
Scheduled moves 0, completed moves 0, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
=== Heat ===
Bed heaters = 0 -1 -1 -1, chamberHeaters = -1 -1 -1 -1
Heater 0 is on, I-accum = 0.3
Heater 1 is on, I-accum = 0.5
=== GCodes ===
Segments left: 0
Movement lock held by File
HTTP is idle in state(s) 0
Telnet is idle in state(s) 0
File is doing "G4 S30                                                       " in state(s) 0
USB is idle in state(s) 0
Aux is idle in state(s) 0
Trigger is idle in state(s) 0
Queue is idle in state(s) 0
LCD is idle in state(s) 0
Daemon is idle in state(s) 0
Autopause is idle in state(s) 0
Code queue is empty.
=== Network ===
Slowest loop: 64.49ms; fastest: 0.00ms
Responder states: HTTP(0) HTTP(0) HTTP(0) HTTP(0) FTP(0) Telnet(0), 0 sessions
HTTP sessions: 1 of 8
- WiFi -
Network state is active
WiFi module is connected to access point 
Failed messages: pending 0, notready 0, noresp 0
WiFi firmware version 1.25
WiFi MAC address 10:52:1c:e1:65:d1
WiFi Vcc 3.40, reset reason Power up
WiFi flash size 4194304, free heap 18720
WiFi IP address 192.168.0.212
WiFi signal strength -41dBm, mode 802.11n, reconnections 0, sleep mode modem
Clock register 00002002
Socket states: 0 0 0 0 0 0 0 0
```

WheelerCC, 2021-06-06:
Just to drop some further info, the number of reconnects when just idle sitting on the home page has significantly increased as well. 

[image]

chrishamm, 2021-06-07:
I cannot reproduce this with Firefox or Chromium. What OS/WiFi router do you have? Does the same problem persist when you're connected from a different device? Did you follow [this](https://duet3d.dozuki.com/Wiki/WiFi_disconnections_and_AJAX_timeout_errors) guide?

chrishamm, 2021-06-15:
Closing due to inactivity.

closed 2021-06-15 by chrishamm