Duet3D/DuetWebControl #381: Input Shaping plugins

mfs12, 2021-08-11:
### Is your feature request related to a problem? Please describe.

Ex. I'm always frustrated when...

### Describe the solution you propose.

Ex. It would be great if...

### Describe alternatives you've considered

Ex. A workaround exists but it is tedius, for example...

### Provide any additional context or information.

Ex. Photos, mockups, etc.

chrishamm, 2021-08-11:
Some thoughts:
- [ ] show effective parameters of selected shaper on final page, see table on https://www.klipper3d.org/Resonance_Compensation.html#technical-details
- [ ] plot graphs for each IS on frequency chart -> https://github.com/KevinOConnor/klipper/blob/master/scripts/graph_shaper.py
- [ ] hide frequencies up to 10(?) Hz and don't use 1Hz for auto-fill

mfs12, 2021-08-11:
- Shouldn't there also one or more frequencies to avoid per axis?

mfs12, 2021-08-12:
@chrishamm
>  hide frequencies up to 10(?) Hz and don't use 1Hz for auto-fill

To gather more experience on this topic i recommend not to hide things in this early stage. But for future less experimental releases this might make definitely sense.

chrishamm, 2021-08-12:
Well, I don't mind about the chart display but it would be definitely good to omit frequencies lower than 10 Hz for the auto-fill function. RRF's firmware limit (at least for the Duet Mini 5+) for the lowest possible IS frequency is around 6 Hz. Klipper's documentation doesn't mention any reasonable values lower than 30 Hz either.

mfs12, 2021-08-12:
- take the tool number into account if there is more than one tool.
- maybe have a data entry field for that, defaulting to the current tool if there is one.
- then the tool number should be included in the filename

mfs12, 2021-08-17:
@dc42 thinking about displaying more then one graph i am not so sure how well this can be managed. Because things will get very quickly crowded as there are 3 graphs (x, y, z) per profile. So displaying 2 profiles will end up with 6 graphs. If the profile and the FFT is displayed it's already 12 in one chart.

Does it makes sense to display only the recorder profile axis? Or give each profile a separate set of graphs?

chrishamm, 2021-08-17:
Since we select a single axis on the first page, I think it's safe to hide the others when a file is automatically loaded. We only need to worry about the selected one.

mfs12, 2021-08-17:
## more specfication from meeting with @dc42
- introduce something like a session
  - configure movement parameters once
  - movement parameters should not be asked all the time
  - allow multiple profile runs per session
- displaying graphs of profile and its fft
  - show only one axis at a time
    - checkboxes to enable axis which to show
  - allow the user to switch between axis
  - allow displaying up to four recorded profiles
    - checkboxes to enable files to show
  - store files of a session in a subfolder?
- display current inputshaping configuration
- next steps (need more specification)
  - save parameters to configuration using M500?
  - add support to configure input shaping per axis?

dc42, 2021-08-18:
> Since we select a single axis on the first page, I think it's safe to hide the others when a file is automatically loaded. We only need to worry about the selected one.

I agree, mostly we are only interested in the axis that was moved, so we can display just that axis by default. However, it should be possible to see the analysis for other axes too, but not necessarily all at the same time.

mfs12, 2021-08-19:
User SIam wrote in the forum https://forum.duet3d.com/post/246038

> After click on the "Record Profile" can you show a message box with collecting data / processing data ... I always forget that the plugin works in the background

mfs12, 2021-09-16:
- [ ] change default damping to 0.1 (according to klipper it's a good default)
- [ ] display records with their test parameters in chart view
- [ ] calculate shaper parameters, refer to https://www.klipper3d.org/Resonance_Compensation.html#technical-details
- [ ] add storage (either local browser storage or file storage)
- [ ] suggest parameters for settings
  - don't use frrequencies below 7 Hz
- [ ] show busy indicator when data is recorded

dc42, 2021-10-15:
It's also unlikely that frequencies above ~80Hz are genuine resonances. One use has a spike at ~120Hz and it turned out to be vibration from a fan. Maybe we should do an accelerometer run with the print head at the usual final position but without any motion, to get a baseline.

mfs12, 2021-10-30:
- [ ] make deletion of recorded files configurable

mfs12, 2021-11-05:
- [ ] implement locking of session data, with vuex?

chrishamm, 2023-03-29:
The IS plugin is now part of DWC.