Duet3D/DuetWebControl #399: [FeatureRequest]: Webcam: stop streaming if not visible

NikAWing, 2021-12-27:
### Is your feature request related to a problem? Please describe.

It looks like the webcam streams even if it is not visible in the webcam viewer of DWC
I see a constant downstream the moment I open the webcam viewer in DWC and it won't stop if I go back to dashboard, status etc.

### Describe the solution you propose.

I think it would save some resources if it would stop streaming while being invisible :)

### Describe alternatives you've considered

none

chrishamm, 2022-01-28:
Good idea, this will be implemented in the next version.

chrishamm, 2022-02-10:
Now implemented in 3.4-rc1.

chrishamm, 2022-02-18:
The corresponding `img` tag is effectively destroyed when you go to another page, so I don't see why the stream would remain active. What browser are you using and would it be possible to use another one for testing purposes?

NikAWing, 2022-02-18:
I notice it when opening the webcam page, it takes a moment to init the stream
I only have (newest) firefox, I might be able to test it on tablet with chrome, but I can not do that today though
I will also try to let FF "forget" the website data and see if that changes something (not sure if that even is related to the problem I have)
I'll comment again as soon as I have results :)

NikAWing, 2022-02-20:
Alright, just made some tests.
PC, Firefox and "forget website", connecting to printer, webcam page, dashboard page (which BTW seems to always take several secons to load), back to webcam page and twice the bandwidth used
then to status page and back to webcam page, thrice the bandwidth used

tried it on smartphone with firefox browser, same behavior.
and also the same with chrome browser on smartphone

chrishamm, 2022-02-22:
I'll attempt to fix this in the upcoming 3.4-rc2.