Duet3D/DuetWebControl #409: [FeatureRequest]:

vaderj, 2022-02-27:
### Is your feature request related to a problem? Please describe.

Ex. I'm always frustrated when...
When you are updating your config files, it would be nice to be able to view prior versions.  

Sometimes it seems that config changes take a few iterations to become effective

### Describe the solution you propose.

Ex. It would be great if...
It sure would be dandy if there was a way to sync the config with stuff like github/onedrive/google drive/(s)ftp(s) and it just maintained a date-based naming schema to avoid namespace collision

### Describe alternatives you've considered

I guess an alternative is manually backing up changed files but I forget to do it all the time

puddly, 2022-03-03:
If this is out of scope for DWC, a workflow that I have found useful is to initialize an empty Git repository and sync files back and forth with https://github.com/adammhaile/rrgit/.

You can then edit config files in a standard text editor and sync the changes.

chrishamm, 2022-08-03:
There was a bug possibly affecting the correct saving of config.g in previous firmware versions but it should be fixed in v3.4.1. If there is another problem, please open up a new thread with more details.

closed 2022-08-03 by chrishamm