Duet3D/DuetWebControl #418: [Bug]:

Juslee73, 2022-04-27:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/28326/m584-help-needed/5

### Which Duet products are you using?

- [ ] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [X] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [ ] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

3.3

### Duet Web Control Version

Dwc3.4

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122 Report
Show all drivers acting normal
```

### Please upload the content of your config.g file.

```gcode
Config.g
Under drivers
All parameters listed correctly 

M584 X0.0:0.1 Y0.2 Z0.4:0.5 E0.3
```

### Details specific to your printer.

Custom core xy nema 23s with dual x and dual z on lead screw 

### What happened?

Expected result when I pressed home all… all 3 axes jerked then the home x home y and home z buttons turned blue. I then tried to move the y axis 10 

Observed result Y axes moved 10 but so does the z axis. It’s like the m584 command is not setting up correctly on the board or the bug could be in the dwc. Why would y&x move whyen I press the move y button on the dwc dashboard????

Steps to reproduce
I have no idea

Juslee73, 2022-04-27:
I will upload the config.g file tomorrow

x0rtrunks, 2022-04-27:
Let's continue troubleshooting in the forum first. Thanks.

closed 2022-04-27 by x0rtrunks