Duet3D/DuetWebControl #422: [Bug]:

hrvojekern, 2022-06-12:
### Duet Forum Discussion Thread

https://forum.duet3d.com/topic/28851/heightmap-in-dwc-can-t-rotate

### Which Duet products are you using?

- [X] Duet2-Wifi
- [ ] Duet2-Ethernet
- [ ] Duet Expansion Breakout Board
- [ ] Duex2
- [ ] Duex5
- [ ] Duet2-Maestro
- [ ] Maestro Dual Driver Expansion
- [ ] Duet3-6HC
- [ ] Duet3-3HC
- [ ] Duet3-1XD
- [ ] Duet3-1LC
- [ ] Duet3-Tool Distribution Board
- [ ] Duet3-Mini5+
- [ ] Duet3-Mini2+
- [ ] Raspberry Pi or other SBC
- [ ] SmartEffector
- [ ] Magnetic Filament Sensor
- [ ] Laser Filament Sensor
- [ ] PT100 Daughterboard
- [ ] Thermocouple Daughterboard
- [X] PanelDue
- [ ] Other
- [ ] None

### Firmware Version

RepRapFirmware 3.4.1 PanelDuefirmware 3.4.0

### Duet Web Control Version

DWC 3.4.1

### Are you using a Single Board Computer (RaspberryPi) with your Duet?

- [ ] Yes I use a SBC.
- [X] No I do not use a SBC.

### Please upload the results of sending M122 in the gcode console.

```shell
M122
=== Diagnostics ===
RepRapFirmware for Duet 2 WiFi/Ethernet version 3.4.1 (2022-06-01 21:05:28) running on Duet WiFi 1.02 or later
Board ID: 08DGM-956GU-DJ3S0-7JKF4-3S86P-KBF7G
Used output buffers: 3 of 26 (22 max)
=== RTOS ===
Static ram: 23860
Dynamic ram: 75304 of which 44 recycled
Never used RAM 12872, free system stack 180 words
Tasks: NETWORK(ready,12.5%,242) HEAT(notifyWait,0.0%,333) Move(notifyWait,0.0%,363) MAIN(running,87.4%,442) IDLE(ready,0.1%,30), total 100.0%
Owned mutexes: WiFi(NETWORK)
=== Platform ===
Last reset 00:26:47 ago, cause: power up
Last software reset at 2022-06-12 00:36, reason: User, GCodes spinning, available RAM 8708, slot 2
Software reset code 0x0003 HFSR 0x00000000 CFSR 0x00000000 ICSR 0x0041f000 BFAR 0xe000ed38 SP 0x00000000 Task MAIN Freestk 0 n/a
Error status: 0x00
Aux0 errors 0,0,0
Step timer max interval 0
MCU temperature: min 28.4, current 37.6, max 38.0
Supply voltage: min 24.8, current 25.0, max 25.2, under voltage events: 0, over voltage events: 0, power good: yes
Heap OK, handles allocated/used 0/0, heap memory allocated/used/recyclable 0/0/0, gc cycles 0
Events: 0 queued, 0 completed
Driver 0: standstill, SG min n/a
Driver 1: standstill, SG min n/a
Driver 2: standstill, SG min n/a
Driver 3: standstill, SG min n/a
Driver 4: standstill, SG min n/a
Driver 5: 
Driver 6: 
Driver 7: 
Driver 8: 
Driver 9: 
Driver 10: 
Driver 11: 
Date/time: 2022-06-12 14:41:37
Cache data hit count 4294967295
Slowest loop: 96.60ms; fastest: 0.18ms
I2C nak errors 0, send timeouts 0, receive timeouts 0, finishTimeouts 0, resets 0
=== Storage ===
Free file entries: 10
SD card 0 detected, interface speed: 20.0MBytes/sec
SD card longest read time 2.6ms, write time 0.0ms, max retries 0
=== Move ===
DMs created 83, segments created 0, maxWait 0ms, bed compensation in use: none, comp offset 0.000
=== MainDDARing ===
Scheduled moves 0, completed 0, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
=== AuxDDARing ===
Scheduled moves 0, completed 0, hiccups 0, stepErrors 0, LaErrors 0, Underruns [0, 0, 0], CDDA state -1
=== Heat ===
Bed heaters 0 -1 -1 -1, chamber heaters -1 -1 -1 -1, ordering errs 0
=== GCodes ===
Segments left: 0
Movement lock held by null
HTTP is idle in state(s) 0
Telnet is idle in state(s) 0
File is idle in state(s) 0
USB is idle in state(s) 0
Aux is idle in state(s) 0
Trigger is idle in state(s) 0
Queue is idle in state(s) 0
LCD is idle in state(s) 0
Daemon is idle in state(s) 0
Autopause is idle in state(s) 0
Code queue is empty
=== Network ===
Slowest loop: 15.67ms; fastest: 0.00ms
Responder states: HTTP(0) HTTP(0) HTTP(0) HTTP(0) FTP(0) Telnet(0), 0 sessions
HTTP sessions: 1 of 8
- WiFi -
Network state is active
WiFi module is connected to access point 
Failed messages: pending 0, notready 0, noresp 0
WiFi firmware version 1.26
WiFi MAC address 8c:ce:4e:f4:56:dc
WiFi Vcc 3.43, reset reason Turned on by main processor
WiFi flash size 4194304, free heap 25040
WiFi IP address 192.168.5.10
WiFi signal strength -48dBm, mode 802.11n, reconnections 0, sleep mode modem
Clock register 00002002
Socket states: 0 0 0 0 0 0 0 0
```

### Please upload the content of your config.g file.

```gcode
; Configuration file for Duet WiFi (firmware version 3.3)
; executed by the firmware on start-up
;
; generated by RepRapFirmware Configuration Tool v3.3.5 on Thu Oct 28 2021 15:34:12 GMT+0200 (Central European Summer Time)

; General preferences
M575 P1 S1 B57600                              ; enable support for PanelDue
G90                                            ; send absolute coordinates...
M83                                            ; ...but relative extruder moves
M550 P"VORON"                                  ; set printer name
M669 K2 X1:0:1 Y0:1:0 Z1:0:-1                  ; select CoreXZ mode

; Network
M552 S1                                      ; enable network
M586 P0 S1                                     ; enable HTTP

; Drives
M569 P0 S1                                     ; physical drive 0 goes forwards
M569 P1 S1                                     ; physical drive 1 goes forwards
M569 P2 S1                                     ; physical drive 2 goes forwards
M569 P3 S1                                     ; physical drive 3 goes forwards
M584 X0 Y1 Z2 E3                               ; set drive mapping
M350 X16 Y16 Z16 E16 I1                            ; configure microstepping with interpolation
M92 X80.00 Y80.00 Z80.00 E427.45               ; set steps per mm
M566 X600.00 Y600.00 Z600.00 E300.00           ; set maximum instantaneous speed changes (mm/min)
M203 X30000.00 Y30000.00 Z30000.00 E3600.00      ; set maximum speeds (mm/min)
M201 X3000.00 Y3000.00 Z3000.00 E7000.00          ; set accelerations (mm/s^2)
M906 X1300 Y1300 Z1300 E600 I30                ; set motor currents (mA) and motor idle factor in per cent
M84 S30                                        ; Set idle timeout

; Axis Limits
M208 X0 Y-215 Z-3.0 S1                               ; set axis minima
M208 X250 Y210 Z230 S0                         ; set axis maxima

; Endstops
M574 X2 S1 P"^xstop"                           ; configure switch-type (e.g. microswitch) endstop for high end on X via pin ^xstop
M574 Y2 S1 P"^ystop"                           ; configure switch-type (e.g. microswitch) endstop for high end on Y via pin ^ystop
;M574 Z1 S2 P5                                  ; configure Z-probe endstop for low end on Z

; Z-Probe
M558 P5 C"zprobe.in" H9 F400 T6000            ; set Z probe type to unmodulated and the dive height + speeds
G31 P500 X0 Y25 Z1.84
                         ; set Z probe trigger value, offset and trigger height
M557 X25:270 Y10:210 S100                         ; define mesh grid

; Heaters
M308 S0 P"bedtemp" Y"thermistor" T100000 B4138 ; configure sensor 0 as thermistor on pin bedtemp
M950 H0 C"bedheat" T0                          ; create bed heater output on bedheat and map it to sensor 0
M307 H0 B0 S1.00                               ; disable bang-bang mode for the bed heater and set PWM limit
M140 H0                                        ; map heated bed to heater 0
M143 H0 S120                                   ; set temperature limit for heater 0 to 120C
M308 S1 P"e0temp" Y"thermistor" T100000 B4138  ; configure sensor 1 as thermistor on pin e0temp
M950 H1 C"e0heat" T1                           ; create nozzle heater output on e0heat and map it to sensor 1
M307 H1 B0 S1.00                               ; disable bang-bang mode for heater  and set PWM limit
M143 H1 S280                                   ; set temperature limit for heater 1 to 280C
M308 S10 Y"mcu-temp" A"MCU"
M308 S11 Y"drivers" A"Duet stepper drivers"

; Fans
M950 F0 C"fan0" Q500                           ; create fan 0 on pin fan0 and set its frequency
M106 P0 S0 H-1                                 ; set fan 0 value. Thermostatic control is turned off
M950 F1 C"fan1" Q500                           ; create fan 1 on pin fan1 and set its frequency
M106 P1 S1 H1 T45                              ; set fan 1 value. Thermostatic control is turned on
M950 F2 C"fan2" Q500
M106 P2 T45:65 H10:11

; Tools
M563 P0 D0 H1 F0                               ; define tool 0
G10 P0 X0 Y0 Z0                                ; set tool 0 axis offsets
G10 P0 R0 S0                                   ; set initial tool 0 active and standby temperatures to 0C

; Custom settings are not defined
M593 P"zvddd" F40.0  ; use ZVDDD input shaping to cancel ringing at 40.0Hz

M572 D0 S0.025				 	; Pressure advance

; Miscellaneous
M501                                           ; load saved parameters from non-volatile memory
```

### Details specific to your printer.

Voron switchwire corexz

### What happened?

heightmap In DWC can’t rotate since firmware version 3.4 and the latest 3.4.1 on tablet (iPad, and iPhone on latest iOS). It was working on 3.3 version.
I would downgrade but want to use input shaping with aceelometer (Duet 2 WIFI rrf 3.4.1)

hrvojekern, 2022-06-12:
heightmap In DWC can’t rotate since firmware version 3.4 and the latest 3.4.1 on tablet (iPad, and iPhone on latest iOS). It was working on 3.3 version.
I would downgrade but want to use input shaping with aceelometer (Duet 2 WIFI rrf 3.4.1)

hrvojekern, 2022-06-12:
sorry just checked ipad ios 12.5.5(ipad air) old but still using it on ios 15.4.1 rotation is working(on height map) in DWC 3.4.1

chrishamm, 2023-01-05:
I'm sorry but we won't support old browsers much longer. Please use the newer iOS then.

closed 2023-01-05 by chrishamm