Duet3D/DuetWebControl #427: [FeatureRequest]:

MintyTrebor, 2022-08-14:
### Is your feature request related to a problem? Please describe.

It might be helpful if DWC did not effectively block the whole web interface when it looses connection to DSF/RRF. This denies access to DWC features that may still be usable (especially plugins & webcams).

### Describe the solution you propose.

Grey-out/disable features that should not be accessible if dsf/rrf is restarting/stopped/powered off (instead of full screen "connection lost"), and display a connection status notice in the top application bar, and include a global flag for plugins that allows them to specify if they should be enabled/disabled during a disconnection event. - DWC would then function more like a web app with limited offline capabilities.

### Provide any additional context or information.

Example Scenario: A Smart Power Socket which is used to power the duet board on/off. If BtnCmd was accessible/functional in DWC when the board is powered off, then it still would be capable of sending the "turn on" command to the smart power socket, as its functions are all browser based client side.

chrishamm, 2023-01-05:
How is DWC supposed to be reachable in the first place if the Duet is turned off? If the web server is not reachable, it may lead to other issues with plugins, so I am not keen on changing this. For this particular case it may make more sense to use external +5V supply so the web server keeps running and the Duet/SBC remains responsive.

MintyTrebor, 2023-01-05:
Agreed - this would only work if you already had the web app loaded in the browser in its current form.
I was thinking about DWC as a PWA - where it could be loaded locally, and still have access to selected features (that were board independant) even if the board was not turned on.

chrishamm, 2023-01-05:
Is there any way to detect if it is running as a PWA? If yes, I could make the "Connecting" dialog dismissable in PWA mode like when the machine is in "Halted" state.

MintyTrebor, 2023-01-05:
My limited understanding is PWA mode availability has to be explicitly defined/included as part of the vuetify project to make the option to "install as application" available to the end user. 

A quick search suggests something along the lines of 

`if (window.matchMedia('(display-mode: standalone)').matches) {
  console.log("This is running as standalone.");
}`

Is one method of detecting current operating mode.

chrishamm, 2023-01-05:
OK thanks, I'll add that as a condition then.

closed 2023-01-05 by chrishamm