Duet3D/DuetWebControl #434: Sensor display layout "jumping" around

kuon, 2022-10-30:
In the web UI, the layout is dynamic and it adapts itself to the number of fans and sensors. If there is too many items for a single line, it is displayed on two lines.

On my printer, I happen to be at the limit between one and two lines, and as the fan speed change the layout changes. I have low speed fans for the water cooling radiator, and they run at about 1000rpm, but sometimes 9xx rpm which causes the text length to vary slightly and it is enough to make the layout "jump" between one and two lines display.

As you can see on the screenshots below, the layout changed from one to two lines. The UI keeps switching between the two layout every second or so, and it is quite disturbing. 

[image]

[image]

I propose the following fix, if there is > 4 fans, put the fan status on its own line at the bottom, and it would also help to put numbers, like `1: xxxx, 2: xxxxx, 10: xxxxx` as it is harder to guess which is which when there are more of them.

chrishamm, 2022-10-31:
We will change v3.5 to only report fan RPM for fans that actually have a tacho port configured. That should reduce the cell width as well. I don't think you have 6 fan tacho inputs configured, do you?

chrishamm, 2022-10-31:
Actually it looks like we already report -1 for fans that do not have a tacho port configured and do not display them on DWC in that case. Are you using version v3.4.4?

kuon, 2022-10-31:
I do use 3.4.4 and all those fans have tacho input, I'd like to be able to monitor their speed because it is a large water cooled machine, and a failing fan could cause significant damage.