Duet3D/DuetWebControl #452: "sbcPythonDependencies" should not fail on builtin modules

stuartofmt, 2023-03-31:
### Is your feature request related to a problem? Please describe.

The current behavior is that if a python module is cited in "sbcPythonDependencies" and that module is a builtin, the install of the plugin fails.

### Describe the solution you propose.

It would be nicer if the installer simply bypassed inbuilt modules.

### Describe alternatives you've considered

The workaround is that the author of a plugin needs to trawl though their import commands and isolate those that are not builtin.

### Provide any additional context or information.

Ex. Photos, mockups, etc.

chrishamm, 2023-04-06:
I don't know Python or its ecosystem, so please provide further details how this can be addressed. At present DSF just invokes pip to install the given dependencies.

stuartofmt, 2023-04-06:
Hopefully this python script is helpful.

I've used this code in the past.  In fact, I commented it out from DuetLapse3, because of the plugin capabilities.
`
#!/usr/bin/python3
import subprocess
import sys

modules = {'platform', 'argparse', 'shlex',"pandas"}   # etc etc etc

for m in modules:
        print('checking: ' + m)
        try:
                globals()[m] = __import__(m)
                print(m + ' is already available')
        except ImportError:
                print('Trying to install: ' + m)
                cmd = 'pip3 install ' + m
                subprocess.run(cmd , shell=True)
        finally:
                if m not in sys.modules:
                        try:
                                globals()[m] = __import__(m)
                        except ImportError:
                                print('Could not import: ' + m)
`

chrishamm, 2023-04-18:
I'd be happy to ship an installer script for Python modules with DSF but this looks just like an example file. If you are willing to provide a robust solution, I'll try to add that to the next v3.5 beta.

stuartofmt, 2023-04-18:
No Problem, can do.  A couple of questions.
1. I assume you call the installer script as a OS level command and make one call per module?
2. What is the format for specifying the module?  I'm guessing :  SomePackage==1.0.4` or  `SomePackage>=1.0.4` or `SomePackage`
3. Is there value in distinguishing between exit code 0 (successfully installed) and exit code x (already installed) and exit code y (failed to install )?  or just keep it a success / fail ?

So basically (on your side) iterate through the dependencies and call, for example, 
 `python_module_installer  "SomePackage==1.0.4"`

chrishamm, 2023-04-18:
Each Python package is installed one-by-one, see https://github.com/Duet3D/DuetSoftwareFramework/blob/v3.5-dev/src/DuetPluginService/Commands/InstallPlugin.cs#L324

I guess it supports whatever pip does, the current install commandline is `pip3 install <name>`.

There is no special handling non-zero return codes yet.

stuartofmt, 2023-04-18:
Thx - I know nothing of c# but enough of "stuff" to set up a test rig using your baseline code.
As soon as I can ....

stuartofmt, 2023-04-22:
I have the script running (with a few embelishments) and tested on 3.4.5.  Along the way it occured to me that we need a "policy" decision.  Since the python environment is (I believe) shared among plugins:  one plugin could over-ride version levels requested by another.
To guard against this, I suggest that the policy is basically:  Rule1: If a newer version of a python module is requested, install the newer version, else stick with the current version. Rule 2 If an older version is requested and a newer version is present, ignore the request.
As well, I'd suggest restricting version requests to == and >== and <= and ~= .  Pip supports max / min ranges and the like but all that makes Rules 1 and 2 difficult to evaluate.

Thoughts ?

stuartofmt, 2023-04-25:
I've gone ahead and implemented the suggestion above and tested on 3.4.5.  Works well (or I least I have not broken it yet).

The code and some notes are here:
https://github.com/stuartofmt/pip3_install

[pip3_install.zip](https://github.com/Duet3D/DuetWebControl/files/11316026/pip3_install.zip)

chrishamm, 2023-04-26:
Thanks! I'll give it a try.

chrishamm, 2023-04-26:
I just added it to the v3.5-dev branch but I haven't tested it yet.