Duet3D/DuetWebControl #453: Suggested improvement to View DCS Log desktop shortcut

stuartofmt, 2023-03-31:
### Is your feature request related to a problem? Please describe.

The existing View Log desktop shortcut on the Duet3d Pi image uses
sudo /usr/bin/journalctl -u duetcontrolserver -e

This form does not wrap the output lines and makes it difficult to see everything especially if using a remote viewer (NoMachine in my case)

### Describe the solution you propose.

The following forms may provide a better experience as they wrap lines
sudo SYSTEMD_LESS=RXMK /usr/bin/journalctl -u duetcontrolserver -e
or
sudo SYSTEMD_LESS=RXMK /usr/bin/journalctl -u duetcontrolserver -n 100

### Describe alternatives you've considered

Ex. A workaround exists but it is tedius, for example...

### Provide any additional context or information.

Ex. Photos, mockups, etc.

chrishamm, 2023-04-06:
Thanks, that's a good idea. `sudo` isn't required on the Pi but I'll give those `less` arguments a try.

stuartofmt, 2023-04-06:
sudo is a bit of a habit for me with some commands :-) 

Other interesting  possibilities (and sometimes more compact output) can be had using cat and / or  grep instead of less.  In some situations it can give a bit more flexibility.

# tail from last boot
journalctl -u duetcontrolserver -b -o cat --no-pager -f

# recent entries from last boot for non unit process 
journalctl -b -o cat --no-pager -e | grep -i duetlapse3